OrebroUniversity/yumi #5: yumi_description xacro parsing error on Ros Kinetic

neckutrek, 2016-12-08:
Hi,

Upon launching yumi_rviz.launch in kinetic the following happens:

    hiqp2016@AASS-CADA06:~/ros_workspace$ roslaunch yumi_description yumi_rviz.launch
    ... logging to /home/hiqp2016/.ros/log/97df3a0c-bd24-11e6-949d-000ffed19a26/roslaunch-AASS-CADA06-17359.log
    Checking log directory for disk usage. This may take awhile.
    Press Ctrl-C to interrupt
    Done checking log file disk usage. Usage is <1GB.

    xacro: Traditional processing is deprecated. Switch to --inorder processing!
    To check for compatibility of your document, use option --check-order.
    For more infos, see http://wiki.ros.org/xacro#Processing_Order
    deprecated: xacro tags should be prepended with 'xacro' xml namespace.
    Use the following script to fix incorrect usage:
            find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'
    when processing file: /home/hiqp2016/ros_workspace/src/yumi/yumi_description/urdf/yumi.urdf.xacro

    xacro.py is deprecated; please use xacro instead
    started roslaunch server http://AASS-CADA06:33801/

    SUMMARY
    ========

    PARAMETERS
     * /joint_state_publisher/use_gui: True
     * /robot_description: <?xml version="1....
     * /rosdistro: kinetic
     * /rosversion: 1.12.6

    NODES
      /
        joint_state_publisher (joint_state_publisher/joint_state_publisher)
        robot_state_publisher (robot_state_publisher/state_publisher)
        rviz (rviz/rviz)

    auto-starting new master
    process[master]: started with pid [17374]
    ROS_MASTER_URI=http://localhost:11311

    setting /run_id to 97df3a0c-bd24-11e6-949d-000ffed19a26
    process[rosout-1]: started with pid [17387]
    started core service [/rosout]
    process[joint_state_publisher-2]: started with pid [17398]
    process[robot_state_publisher-3]: started with pid [17405]
    process[rviz-4]: started with pid [17424]
    [rviz-4] process has finished cleanly
    log file: /home/hiqp2016/.ros/log/97df3a0c-bd24-11e6-949d-000ffed19a26/rviz-4*.log`

gavanderhoorn, 2016-12-08:
Seems like it works correctly, no? The `xacro` message is just a warning at this point.

tstoyanov, 2016-12-20:
I also think we can live with this warning. Changing to the new xacro will probably break backwards compatibility, so let's wait for a more pressing reason to do it.

closed 2016-12-20 by tstoyanov