OrebroUniversity/yumi #6: Yumi's publication with config file

juliamp22, 2016-12-13:
Hi to all,
I'm trying to publish to real and robotstudio yumi's robot with the topics of /joint_path_command  with a config file that contained all the joints of the robot in order to change the real joints at the same time I change the rqt_reconfigure. It works the first time, but then, it appears the next message:

Validation failed: Missing valid timestamp data for trajectory pt 1

The clock of the robot and the computer are the same, so I think that it has to be from de ros_motion program. 

In order to this, the buffer of the publisher is 10 and the ros loop works every 0.48 seconds.

Thanks for all,

Júlia

tstoyanov, 2016-12-20:
Hi Julia,

could you please provide some more details on your setup for replication purposes?
What branch of the repository are you using? What is running on the yumi controller side? What is running on the ros side?

The set up on the side of RAPID should be done almost identically to the setup for the abb_driver in ros_industrial. I was planning to write a short how-to to guide people through, but I am not sure when I will have the time to get to it...

juliamp22, 2017-02-23:
It was my fault, I was trying to publish without a duration. 

Thans for all,

Júlia

tstoyanov, 2017-02-28:
@juliamp22 can I close this issue then?

juliamp22, 2017-03-01:
Yes, you should.

Thanks for all,

Júlia

closed 2017-03-01 by tstoyanov

nali12, 2019-06-11:
Hello juliamp22, How did you solve this issue?