OrebroUniversity/yumi #8: Error with lauch file in gazebo

glebysg, 2017-02-23:
When I try `rosrun gazebo_ros gazebo` on yumi_hw.lauch or  yumi_gazebo_pos.launch
I'm getting the following error:

```
Gazebo multi-robot simulator, version 2.2.3
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 2.2.3
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Msg Waiting for master.[ INFO] [1487821532.561283037]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1487821532.561655933]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Msg Waiting for master
Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 128.46.125.51
Error:   Could not find the 'robot' element in the xml file
         at line 81 in /build/buildd/urdfdom-0.2.10+dfsg/urdf_parser/src/model.cpp
Error [parser_urdf.cc:2608] Unable to call parseURDF on robot model
Error [parser.cc:273] parse as old deprecated model file failed.
Error [Server.cc:300] Unable to read sdf file[/home/glebys/catkin_ws/src/yumi/yumi_launch/launch/yumi_gazebo_pos.launch]
gzserver: /usr/include/boost/thread/pthread/recursive_mutex.hpp:101: boost::recursive_mutex::~recursive_mutex(): Assertion `!pthread_mutex_destroy(&m)' failed.
```
I'm using ROS indigo.

tstoyanov, 2017-02-23:
Hi,

have you installed the ros_control plugins for gazebo? You can do that by running 
`sudo apt-get install ros-indigo-gazebo-ros-control`

glebysg, 2017-02-23:
Thank you, the robot is spanning properly in gazebo.
I'm getting the following error (maybe due to the fact that I'm not calling any additional services)

```
SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: True
 * /yumi/joint_state_controller/publish_rate: 500
 * /yumi/joint_state_controller/type: joint_state_contr...
 * /yumi/joint_trajectory_pos_controller/joints: ['yumi_joint_1_l'...
 * /yumi/joint_trajectory_pos_controller/type: position_controll...
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_1_l/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_1_l/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_1_l/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_1_l/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_1_r/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_1_r/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_1_r/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_1_r/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_2_l/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_2_l/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_2_l/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_2_l/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_2_r/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_2_r/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_2_r/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_2_r/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_3_l/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_3_l/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_3_l/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_3_l/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_3_r/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_3_r/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_3_r/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_3_r/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_4_l/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_4_l/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_4_l/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_4_l/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_4_r/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_4_r/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_4_r/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_4_r/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_5_l/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_5_l/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_5_l/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_5_l/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_5_r/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_5_r/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_5_r/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_5_r/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_6_l/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_6_l/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_6_l/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_6_l/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_6_r/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_6_r/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_6_r/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_6_r/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_7_l/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_7_l/i: 0.002
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_7_l/i_clamp: 0.1
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_7_l/p: 1.5
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_7_r/d: 0
 * /yumi/joint_trajectory_vel_controller/gains/yumi_joint_7_r/i: 0.002
[... 45 lines cut ...]

Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 128.46.125.51
[spawn_urdf-3] process has finished cleanly
log file: /home/glebys/.ros/log/e218fa10-f9fe-11e6-af44-f48e387f42e4/spawn_urdf-3*.log
[ INFO] [1487878936.750763708, 0.365000000]: Loading gazebo_ros_control plugin
[ INFO] [1487878936.750871139, 0.365000000]: Starting gazebo_ros_control plugin in namespace: /yumi
[ INFO] [1487878936.752033381, 0.365000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ INFO] [1487878936.898975857, 0.365000000]: Loaded gazebo_ros_control.
[INFO] [WallTime: 1487878937.086840] [0.549000] Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [WallTime: 1487878937.088817] [0.550000] Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [WallTime: 1487878937.091499] [0.553000] Loading controller: joint_state_controller
[... cut ...]
[ERROR] [1487878938.099645352, 1.554000000]: Could not load controller 'joint_trajectory_pos_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1487878938.099679647, 1.554000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1487878939.100426] [2.538000] Failed to load joint_trajectory_pos_controller
[INFO] [WallTime: 1487878939.100801] [2.538000] Controller Spawner: Loaded controllers: 
[INFO] [WallTime: 1487878939.103902] [2.542000] Started controllers:
```

Thanks!

tstoyanov, 2017-02-28:
Hi,

I think you are still missing packages. Try installing the ros_controllers package. On ubuntu with ros indigo this would be
`sudo apt-get install ros-indigo-ros-controllers`

Verify that you have joint states published afterwards by 
`rostopic echo /yumi/joint_states`

gavanderhoorn, 2017-02-28:
@glebysg: try running:

```
rosdep install --from-paths /path/to/your/catkin_ws/src --ignore-src
```

after you have `source`d the appropriate `setup.bash` (probably the one in `devel/setup.bash`).

That should tell you whether you need to install any additional packages.

glebysg, 2017-04-10:
This worked! 
Thank you very much

closed 2017-04-10 by glebysg