OrebroUniversity/yumi #14: Torque interface

khoramshahi, 2017-06-30:
Hi there,

I just wanted to know if there is an available (plug-and-play) torque interface for Yumi robot.
I would like to send joint torque commands, and also be able to read external force/torques applied to the robot.

Thanks for your help and sorry if this is irrelevant to your repository. 
Mahdi

tstoyanov, 2017-06-30:
Hi Mahdi,

unfortunately that's not a feature we can do anything about at the moment. The interfaces to the robot from the manufacturer's side don't at the moment allow controlling or reading the joint torques. If this changes at some point, we would of course be very eager to add this support :)

T.

closed 2017-06-30 by tstoyanov

khoramshahi, 2017-06-30:
Thanks for your quick response.