OrebroUniversity/yumi #15: Fails to build

stinor, 2017-07-07:
Hi

I have tried to build this following the instructions provided but it fails with the following error

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "abb_egm_interface"
  with any of the following names:

    abb_egm_interfaceConfig.cmake
    abb_egm_interface-config.cmake

I have not been able to find anything related to these in the repository.

Cheers

tstoyanov, 2017-07-07:
Hi,

sorry about that. We are working on some new features which got pushed prematurely into master. I have reverted the merge and you should be able to build now. Can you please check if you are able to build now?

stinor, 2017-07-07:
Hi

Many thanks. I am now able to do a build so will close this issue.

closed 2017-07-07 by stinor