OrebroUniversity/yumi #16: Issue with "yumi_moveit_config/launch/demo_online.launch"

stinor, 2017-07-12:
Hi

We are having the following issue with both the simulated robot in robot studio and the actual yumi robot. When launching the "yumi_moveit_config/launch/demo_online.launch" the following error occurs.

The "right_arm/joint_trajectory_action" and "left_arm/joint_trajectory_action" clients fail to connect as shown in the image below.

[image]

Not quite sure what is wrong here as we have followed the instructions from the wiki and issues list.
We are using the RAPID files from the "src/yumi/yumi_support/rapid" folder as we are trying to control the yumi robot using the moveit rviz gui interface.

tstoyanov, 2017-07-12:
Hi,

I am afraid we might not be able to help you out much. The moveit support is a direct port from the abb_driver in ros-industrial, with very minor modifications. I did that as a first step to developing the ros_control drivers and did not debug it thoroughly. In fact, I am pretty sure the RAPID drivers that you are using would not move all joints of the robot correctly, so be very careful when testing... 

I don't think any of the other maintainers are using moveit to control the robot either, so I think you might need to do some digging and debugging. I remember it sometimes took a very long time to bring up the planning scene, which would mess with the robot controllers. Check if the config uses the full robot meshes or if you are using the coarse meshes. And please post what you find here and I will try to help as I can.

T.

stinor, 2017-07-13:
Hi 

We have managed to get this working now using the two steps below.
We were initially missing step 1 below which is what starts the industrial_robot_client which contains the joint_trajectory_action messaging that was previously missing.

1. From the terminal launch "robot_interface.launch" file as shown below. Ensure the correct robot ip   address is set in this file first.
 "~/catkin_ws/src/yumi/yumi_support/launch$ roslaunch robot_interface.launch"

2. Now in a separate terminal launch the online moveit demo as shown below.
  "~/catkin_ws$ roslaunch yumi_moveit_config demo_online"

We are now able to play around with the moveit.rviz GUI robot and see the actions translated onto the actual yumi robot  and the simulated yumi robot on robot studio.

Cheers

closed 2017-07-13 by stinor