OrebroUniversity/yumi #18: Instructions to load RAPID code to robot studio

neubotech, 2017-07-20:
Hello, great job on getting the yumi interface work with ROS and MOVEIT!  I'm wondering if you have an instruction of how to load your RAPID code to YUMI via robot studio.  Much appreciated!

tstoyanov, 2017-07-21:
Hi,

at the moment there are no instructions here. However, if all you want to do is use YuMi with MoveIt (and not use ros_control), then you can follow the instructions on the abb ros-industrial wiki (http://wiki.ros.org/abb/Tutorials). The steps are the same for YuMi, just use the RAPID files provided in the yumi_support package. You would need to use our fork of ros-industrial (OrebroUniversity/ros-industrial), and you only need to configure one motion/state server for the robot (see #9).  You would still need two different motion execution tasks, one for each arm.

When I have some time I will write a how-to, but for now you might have to figure things out on the go.

YoshuaNava, 2018-01-17:
Hey,
I wrote a guide on how to set the RAPID modules for the different control interfaces: [Firmware](https://github.com/kth-ros-pkg/yumi/wiki/Firmware)

Hope this helps,
Yoshua

closed 2018-01-17 by YoshuaNava