OrebroUniversity/yumi #20: Instructions for setting up grippers in RobotStudio

stinor, 2017-07-25:
Hi

Thank you for your help so far. I have got the two arms working (with robot studio simulated robot and actual robot) and now trying to get the grippers working but I think I'm missing some instructions or some information. Are there any special instructions for setting up the grippers in RobotStudio for this package.

I have loaded the gripper RAPID code from the "yumi/yumi_hw_rapid/GRASP" folder.
There are no errors when I run the "RAPID->Check Program" function in RobotStudio.
However, when I do a warm restart a couple of "Alias Define" errors are reported for the "HandDriver_left.HandSetupChirality()" and "HandDriver_right.HandSetupChirality()" procedures.
These are shown in the picture below.

[image]

**Assumptions**
The error seems to suggest that the aliases for "hand_ActualPosition_L", "hand_ActualSpeed_L", "hand_SysState_L" etc. need to be setup by defining their respective signals through the IO Configuration in RobotStudio. From what I can see, about 20 of these signals may need to be setup for each gripper.

**Question:**

1. Is my assumption above correct and if so are there any instructions or could you point me to anything that will help me with this. I have looked at some tutorials on how to setup IO signals for the gripper (eg. https://www.youtube.com/watch?v=6XoylqyZfFA) but nothing seems close enough to the aliases that are defined here. It seems I may also need to know what to map these aliases to.

2. If my assumption is wrong, could you point me in the right direction. 

 Much appreciated!

tstoyanov, 2017-07-25:
Hi,

afaik, no need to set up signals. At least I never did when I set up the grippers. Maybe @YoshuaNava can also give some input here on his experience of setting up. The hand driver code is a direct copy from an older release of the Smart Gripper robotware package. In principle, if you install the newest version you would not need any of the hand driver modules, but just the ROS_ modules. As the API of Smart Gripper is constantly changing though, you might need to re-name functions. What version of robot ware and smart gripper are you using?

Let me know how you resolve this issue.

stinor, 2017-07-26:
Hi 

I am using RobotWare version 6.05.00.01 and SmartGripper version 3.40

YoshuaNava, 2017-07-27:
Hello,

I will answer your questions below.

## Gripper tasks and modules
I usually set the grippers modules (and all other motion modules) as normal tasks, for safety reasons. Particularly, for using the grippers create the following tasks:

|          Task          |    TYPE    | Trust Level | Use Mechanical Unit Group |
|:----------------------:|:----------:|:-----------:|:-------------------------:|
|  T_GRIP_State          | NORMAL     |   SysFail   |                           |
|  T_GRIP_Motion         | NORMAL     |   SysFail   |                           |
|  T_GRIP_Motion_L       | NORMAL     |   SysFail   |                           |
|  T_GRIP_Motion_R       | NORMAL     |   SysFail   |                           |

After creating them, do a soft restart. Then, load the following modules into each of the tasks listed:

|                     File                     |          Tasks         |          
| -------------------------------------------- |:----------------------:|
| RWS/ROS_common.sys                           |  T_GRIP_State, T_GRIP_Motion, T_GRIP_Motion_L, T_GRIP_Motion_R |
| RWS/ROS_messages.sys                         |  T_GRIP_State, T_GRIP_Motion, T_GRIP_Motion_L, T_GRIP_Motion_R |
| RWS/ROS_socket.sys                           |  T_GRIP_State, T_GRIP_Motion, T_GRIP_Motion_L, T_GRIP_Motion_R |
| Grippers/Gripper_motionServer.mod            |  T_GRIP_Motion         |
| Grippers/Gripper_stateServer.mod             |  T_GRIP_State          |
| Grippers/GripperMotion_left.mod              |  T_GRIP_Motion_L       |
| Grippers/HandDriver_left.sys                 |  T_GRIP_Motion_L       |
| Grippers/GripperMotion_right.mod             |  T_GRIP_Motion_R       |
| Grippers/HandDriver_right.sys                |  T_GRIP_Motion_R       |

The RWS and Grippers folders, containing the RAPID code, are the ones that appear here: 

https://github.com/OrebroUniversity/yumi/tree/feature-egm/yumi_hw/rapid

*Note*: the main feature developed in that branch is the support for Externally Guided Motion (EGM), an interface for real time control. If you clone it and try to build it without the corresponding libraries provided by ABB, it will fail. Hence, if you need the scripts, clone the repo, feature-egm branch, out of your ROS workspace and just access the relevant folder _yumi_hw/rapid/_ to get the scripts.

## Signals

I haven't set any particular set of signals, just worked with the ones that come pre-set in YuMi by default. However, if you do need to set them, you can find a list of them after you unzip any of our firmware backups, as:

https://github.com/OrebroUniversity/yumi/blob/feature-egm/yumi_support/firmware_backups/500693_EGM_AllJoints_BothGrippers_NormalTasks.zip

And checking the file SYSPAR/EIO.cfg

Hope this helps,
Yoshua

PS: I have been writing a tutorial about YuMi firmware setup, but it's not ready yet: 
https://github.com/kth-ros-pkg/yumi/wiki/Firmware-setup

stinor, 2017-07-31:
Hi Yoshua

Thank you for the information above.

I have setup the grippers as shown above using the rapid files from the feature-egm branch.
[https://github.com/OrebroUniversity/yumi/tree/feature-egm/yumi_hw/rapid](url)

I have also set up the signals using a backed up SYSPAR/EIO.cfg file from the [https://github.com/OrebroUniversity/yumi/blob/feature-egm/yumi_support/firmware_backups/500693_EGM_AllJoints_BothGrippers_NormalTasks.zip](url)

This is the procedure I ended up using.
I started from scratch by creating a new yumi controller and did a restore from backup using the backup from the above link. This essentially gave me all the required signals already setup from the SYSPAR/EIO.cfg file.

I then stripped out all the old arm and gripper tasks that came from the backup and manually re-added the arm and gripper tasks as shown above.

Note, that at this stage I have defined all the tasks as semi-static as I am still using the RobotStudio simulator and I want the tasks to start automatically on startup/reset.

When I do a "Rapid->Check Program" everything is ok. Now when I run the simulator, I no longer have the issues I had previously due to missing signal definitions. All tasks startup and run, however after 5seconds the GripperMotion_left and GripperMotion_right tasks both raise an exception. A picture of this is shown below.

[image]

From what I can see, the task is timing out in the Hand_WaitForExpectedState as it is not able to get the Hand_GetFingerState() successfully within the timeout period of 5s. The expectedState passed in is 6 which equates to STATE_ACTIONCOMPLETED. Hand_GetFingerState() returns the hand_SysState so it looks like hand_SysState is not being updated.

I am not sure whats going on beyond this now so again starting to suspect some configuration issue.

Note at this stage, I haven't started any of the ROS nodes on the ubuntu side. I am expecting the RAPID server tasks to be waiting for clients to connect. This is the case for the motion and state servers for the arms and grippers, however the gripper motion task is having this issue.

Would you have an idea of what is going on here.
Many thanks

YoshuaNava, 2017-08-03:
Hello stinor,
I haven't experienced this issue before. Have you been able to solve it?

I have a lot of tasks now, but if you could send me a backup of your current setup, I could test and debug it locally, probably in 1-2 weeks.

stinor, 2017-08-09:
Hi Yoshua

I will send you a backup of my current setup.

I haven't been able to solve this gripper issue yet on the simulator. 
However, everything seems to be working on the real robot which I have been working with this last week. 

Thanks

lz89, 2017-10-23:
@YoshuaNava How could you have the trust level option when using normal mode. In my case the trust level option won't be available if a task is set to normal.

ahmadwahab, 2017-11-19:
I followed the above instructions exactly and I run the RAPID check program I got no errors, however I'm still not able to communicate with the grippers and the when I run the programs Gripper_motionServer and Gripper_stateServer are waiting for connections.
Any help with this issue?