OrebroUniversity/yumi #21: how control the real yumi robot

songcaiwei, 2017-09-07:
hi, I am using the yumi_support and yumi_moveit_config.  when I use the deme_online.launch, I can accept the robot position, but i can't use the motion planning to control the robot move. And I also try use the "rostopic pub -1 /left_arm/joint_path_command trajectory_msgs/JointTrajectory "  to control the arm,  it doesn't work also. 
So, what the problem it is?  
Much appreciated!

tstoyanov, 2017-11-30:
Could you please give some more details about this issue? What branches are you using and what is running on the robot?

LeonRobot, 2017-12-18:
It was my fault, the  set up on the side of RAPID has something wrong.

Thanks for all.

closed 2017-12-18 by LeonRobot

beevabeeva, 2019-11-29:
Having the same issue I think. If you could detail the steps used to solve it (what did you change in the RAPID files?), that would be appreciated.

LeonRobot, 2019-11-30:
> Having the same issue I think. If you could detail the steps used to solve it (what did you change in the RAPID files?), that would be appreciated.
It has been a long time since then, and I have forgotten the problems at that time. This software is no longer maintained by Orebro University.  I used to refer to [kth-ros-pkg/yumi](https://github.com/kth-ros-pkg/yumi/wiki/Firmware) and [ethz-asl/yumi](https://github.com/ethz-asl/yumi/wiki/Setting-Up-YuMi-in-RobotStudio). I hope it will help you

beevabeeva, 2019-12-01:
@LeonRobot thanks. I opened issues on those repos too.