OrebroUniversity/yumi #22: the OrebroUniversity/industrial_core is same as the industrial_core in the ROS-Industrial

songcaiwei, 2017-09-07:
hi, i use the the OrebroUniversity/industrial_core, but when i run the  roslaunch robot_interface.launch 
it show the [ERROR] [1504762335.231789044]: Get unload pointer failed, buffer size: 0, smaller than byte size: 4
[ERROR] [1504762335.231843198]: Unload pointer returned NULL
[ERROR] [1504762335.231877206]: Failed to unload message joint: 8 from data[0]
[ERROR] [1504762335.231913226]: Failed to unload joint data
[ERROR] [1504762335.231945259]: Failed to initialize joint message

BUT if change from  static const industrial::shared_types::shared_int MAX_NUM_JOINTS =20;
 to static const industrial::shared_types::shared_int MAX_NUM_JOINTS =10 ; in simple_message/include/simple_message/joint_data.h ,   It works.

I want to konw that  does the MAX_num_joints=20  is necessary and why the ERRor occer

lz89, 2017-10-23:
I have the same issue.
I though the MAX_NUM_JOINTS should change to 20 for Yumi due to the discussion [here](https://github.com/ros-industrial/industrial_core/issues/147)
What I did was to clone the original industrial_core repo and check out the kinetic stable branch, then change the `MAX_NUM_JOINTS` from 10 to 20.
Then I get the error, but if I change back to 10, the `robot_interface.launch` under yumi_support folder works.

tstoyanov, 2017-10-23:
What branch of the yumi repo are you using? It may be that you have a different value for MAX_NUM_JOINTS on the ros side and the RAPID side. For a good overview on setting things up, I would also suggest checking through the wiki on the kth fork of this repo (https://github.com/kth-ros-pkg/yumi/wiki).

gavanderhoorn, 2017-10-23:
@tstoyanov: the instructions on those pages contain references to the EGM interface driver. Afaik that is (still) not publicly available.

lz89, 2017-10-23:
@tstoyanov I'm using the master branch for ROS side, but feature-egm branch for the RAPID side.
The reason I am doing this is that from the RAPID code under yumi_hw (master branch) only have:
ROS_motion_left.mod and ROS_motion_right.mod. I don't know how should I load these to the tasks: T_ROB_L, T_ROB_R, ROS_MotionServer_L, ROS_MotionServer_R (or only one motion server as you mentioned in another issue that the yumi_hw uses only one motion server for both arm). I did refer the the kth fork which to some extent is helpful but also introduce some confusion as the instruction is not completed and aim for EGM.

I have read through all the opened/closed issues in this repo as well as the ethz fork wiki. I am still confused. So would you please give me some advises about:
1. If I don't have EGM and want to use the direct control (NOT trajectory download), shall I use the yumi_hw under master branch of this repo.
2. If I use yumi_hw master branch, how to link the mod files to the tasks, and what tasks shall I create? I know how to create task and load the modules in the controller etc.
3. If everything is setup at RAPID side (robot), what shall I run in ROS side (PC) in order to establish the communication?
4. I assume I need to use ros_control package from the ROS to control and communicate with the robot? (Sry that I am not that familiar with this)

Thank you very much!

lz89, 2017-10-24:
Regarding the `MAX_NUM_JOINTS` issue, I think I found the answer from:
https://github.com/OrebroUniversity/yumi/pull/19

tstoyanov, 2017-11-30:
Hi @lz89 ,
1. If you want to do direct control and don't have EGM, you should look at the feature-ros-control branch. This does not offer MoveIt support: you would have to use ros_control controller instead (though it is possible to translate to action server, just needs some work)
2. In general, you should load the the ROS_common, ROS_messages and ROS_socket in all tasks (static). You should make a new task under either T_ROB_L or T_ROB_R that loads the ROS_StateServer. You should then load ROS_motion_left to the main task in ROB_L and conversely ROS_motion_right to ROB_R. Loading the gripper tasks should be identical to the instructions on the KTH wiki. 
3. If everything is running smoothly, you should be able to roslaunch yumi_launch yumi_pos_control.launch. This should set up a position streaming interface. I have not tested this extensively. The yumi_vel_control.launch at the moment loads a custom controller that you probably do not have installed. If you modify it to load the joint_trajectory_vel_controller instead, this should give you a nice velocity control interface.
4. Yes. You would need to get  some familiarity there.
5. That should be done by writing a cartesian ros_control controller. There might already be some examples of that available.

LeonRobot, 2017-12-12:
@tstoyanov @lz89 @gavanderhoorn Thank you for your answers. I think this problem has been solved.

closed 2017-12-12 by LeonRobot