OrebroUniversity/yumi #24: Unable to load yumi_moveit_config in mobeit setup asisstant!

janisa9, 2017-09-22:
I don't understand what i am doing wrong but i can't load yumi moveit config in setup asisstant to edit it. 
[image]

Okay setup asisstant says that it is unable to locate urdf file yumi.urdf, although in yumi/description/urdf folder there is not file with such an extension. There are only yumi.urdf.xacro, yumi.xacro and some other files
[image]

Although usually files with urdf.xacro is possible to load in setup asisstant. So can anyone help me to selve this problem or try yourself if you have the same problem?

tstoyanov, 2017-09-25:
Hi,
this is a known issue, as I could not find a way to convince moveit to open the xacro files. Probably there is a better way to do it, but at the moment, I suggest to manually run xacro on the yumi.urdf.xacro, passing along as an argument the interface to the robot that you need (either position or velocity) like this:
`roscd yumi_description/urdf/
rosrun xacro xacro.py yumi.urdf.xacro prefix:=PositionJointInterface > yumi.urdf`

If you come up with a better way (maybe changing something in the setup assistant launch file?) please try it out, test it, and open a pull request. We'd be happy to have this issue solved.

cheers,
T.

gavanderhoorn, 2017-09-25:
A workaround for this could be to create two top-level wrapper xacros that define values for the `prefix` arg. Those could then be loaded into the MSA directly.

Note that on Indigo the MSA can load xacros that require Jade+ xacro since ros-planning/moveit#540. That may further simplify things.