OrebroUniversity/yumi #26: release abb_rws_interface

simonschmeisser, 2017-10-04:
Hi everybody

I noticed some of the newer branches relying on a package called abb_rws_interface. I assume this is a convenience wrapper to the robot web service Rest interface? I was unfortunately unable to get very far in using this interface (I assume I was missing some "log in" step). Do you intend to release this package soon? If not would you mind sharing it privately? Or at least some demo code?

Thanks a lot
Simon

tstoyanov, 2017-11-30:
Hi,
yes, you are correct in assuming that the newer branches (feature-egm-interface) depend on some hidden code. Unfortunately, I cannot tell you when this will be released, as it's not under my control. Maybe @gavanderhoorn has some idea about planned release dates?

gavanderhoorn, 2017-12-01:
At this point I can't give any more info on this.

@simonschmeisser: it will be released, but it's not under my control.

YoshuaNava, 2018-01-17:
@jontje from ABB discussed about this in the following issue:

https://github.com/kth-ros-pkg/yumi/issues/7

closed 2018-01-17 by YoshuaNava