OrebroUniversity/yumi #27: Support for lead-through mode

lz89, 2017-11-02:
Hi,

I wonder if there is a way to enable lead-through mode while the ROS server is running. Therefore, the ROS side can read robot's joint positions without sending command to the robot.
This is particular useful for demonstration as well as hand-eye calibration.

tstoyanov, 2017-11-30:
As far as I know, the answer is no...

YoshuaNava, 2018-01-17:
@lz89 We have been able to do this, thanks to @diogoalmeida.

So, what you should do is:
1. Put the robot in AUTO mode, with motors turned ON.
2. Run the RWS/EGM nodes on your computer.
3. Once the robot starts (the brakes usually make a "click" sound when this happens), change from AUTO to MANUAL mode in the FlexPendant.
4. On the FlexPendant, open the main menu, Jogging. Then, for each arm touch the option "Enable Lead Through"

If everything went well, you should be able to move YuMi's arms without the RWS/EGM connection being lost. Hence, all the joint states should be recordable.

NOTE: Be very careful when you enable lead through. If your robot is carrying any object, its arms will fall. Also, don't move the arms too fast, or you'll get [Motion supervision](https://github.com/kth-ros-pkg/yumi/wiki/Troubleshooting#motion-supervision) triggered.

Please let me know if this helps,

Yoshua

closed 2018-01-17 by YoshuaNava

diogoalmeida, 2018-01-17:
Just note that this is a workaround and not a solution to the problem, and that it works *sometimes*.