OrebroUniversity/yumi #28: Joint velocity setting with yumi_support package

lz89, 2017-11-07:
I am using the yumi_support package and use topic `/xxxx_arm/joint_path_command` for control.
Now I am publishing just one point trajectory to the topic with desired joint values closed to the current value. If I leave the velocity field in the `trajectory_msgs/JointTrajectory` for the trajectory point, the robot moves smooth but slow. When I set the velocity field with a value, the robot moves very slow (slower than without setting it), even with a very large value for the velocity. I started the velocity with 0.1 and increase it to 500000, but there is not much different.

How can I increase the robot velocity when using the `joint_path_command` topic for the robot control?

gavanderhoorn, 2017-11-07:
Cross-posted to ROS Answers: [Velocity setting for topic joint_path_command](https://answers.ros.org/question/275277).

lz89, 2017-11-15:
I tried to modify the default speed in RAPID ([ROS_motion_right.mod](https://github.com/OrebroUniversity/yumi/blob/master/yumi_support/rapid/ROS_motion_right.mod#L39)).
Based on the ABB manual, there are a bunch of predefined speed that can be used:

[image]

I tried from v100 to v7000, the speed seems increase a bit from v100 compared to the original one (v10), but there is not much difference b/w v100 and v7000.
In addition, I tried vmax but an error occurred (not defined variable) when I run the RAPID program.

If you have any experience on this variable please comment what's the preferred and safe value.

tstoyanov, 2017-11-30:
Hi,
I have very little experience with this, but there can be many things at play. Are you running the robot in automatic mode? If you run in manual, it will certainly refuse to go at some of the very high speeds. Another thing you could check is how many trajectory points do you download and how close together they are. If they are closer together than the stop_mode distance, the robot might never have time to accelerate to the desired speed. The final thing I would suggest is to check what trajectory durations you are setting. I have a suspicion that this would actually be the determining factor (while move_speed is just an upper velocity bound).