OrebroUniversity/yumi #29: Gripper Connection Problem

ahmadwahab, 2017-11-23:
Hi,
I was able to install the ABB ros server on my Yumi robot, the server is working good for controlling the two arms using Orebro/yumi package. However I'm still having problems in connecting the grippers to the ROS server. I installed all the required files and modules for grippers control and I run the RAPID check program and I got no errors. The main problem is that the Grippers Motion Server and the Grippers State server are still not connected to the ROS server. Any help with this issue??

tstoyanov, 2017-11-30:
Hi,
I am answering also here for completeness sake. Try to run the yumi_gripper_node executable from the yumi_hw package. This should connect to the RAPID servers and should let you call open/close service calls for each hand.

ahmadwahab, 2017-11-30:
I did so several times with different files from different branches. Even I made an edit in gripper_node.h by try to change the port value but without any benefit. The problem is that the  yumi_gripper_node  is not correctly connected to RAPID servers. That's why I'm not able  to call open/close service calls.
The GRIP_State and GRIP_motion servers are never connected to my ROS PC client even after running the node.

YoshuaNava, 2018-01-17:
Hey,
In the KTH branch we made some modifications to the grippers RAPID modules. Check [here](https://github.com/kth-ros-pkg/yumi/tree/egm_modifications/yumi_hw/rapid/Grippers)

Those modules should be loaded into YuMi internal computer. For a guide on how to do that, see [Grippers and cameras setup](https://github.com/kth-ros-pkg/yumi/wiki/Firmware#grippers-and-cameras)

Hope this helps,
Yoshua

alibinjunaid, 2020-04-03:
@YoshuaNava Hi Yoshua, I have been following the instructions on the wiki page. The arm control is working, however the grippers tasks is giving error with no description when the rapid scripts from the yumi_hw/grippers/* is used. It would be great if you can point if i am something missing. Seems like rapid scripts for gripper seems to have an error. Thank you.