OrebroUniversity/yumi #33: Failed to control IR14000 using ROS/ABB_driver

amranov, 2019-06-21:
Hi guys ,
We are trying to control the real robot with ABB interface we followed this tutorial [https://github.com/kth-ros-pkg/yumi/wiki/Firmware](url) to install the firmware, (We dont have the EGM modules). when we launch : `roslaunch yumi_launch yumi_pos_control.launch` we get this error :
```
[ WARN] [1561104675.666918206]: Send failure, no callback support
[ INFO] [1561104675.666949402]: Connection failed, attempting reconnect
[ERROR] [1561104675.667712026]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
```
Any suggestions?

Sinchiguano, 2019-06-21:
You can try with the following link:
https://github.com/BerkeleyAutomation/yumipy

There my application in pose estimation 
https://github.com/Sinchiguano/3D_Pose_Estimation_YuMiArm_ABB

amranov, 2019-07-11:
@Sinchiguano Thank you

closed 2019-07-11 by amranov