OrebroUniversity/yumi #35: Using Yumi with ROS + MoveIt!

beevabeeva, 2019-11-29:
We have managed to get as far as having MoveIt (+ rviz) get the correct current position of the robot. However, when trying to execute a path to a goal position, the robot does not move.

In addition, there are no gripper topics listed. MoveIT complains about not having the gripper states:
[image]

We followed the instructions (after digging around in other branches to find the required backups (addition-demo, two-grippers realtime) ). We found someone with a similar issue
(https://github.com/OrebroUniversity/yumi/issues/21) but there were no detailed steps listed to solve the issue.

Any help would be appreciated.

Sinchiguano, 2019-11-29:
You can try Yumipy, it is quite easy to move the robot with it, and it is also done in ros environment, so you can use Rviz and everything available from ROS.

beevabeeva, 2019-12-01:
Wr tried installing YumiPy but ran into a lot of issues- we are using ros-kinetic and the YumiPy docs explicitly say it has only been tested on jade. We might give it another go.
Thank you for linking your repo. It looks like it will be helpful. We will try it next week when we return from conference.

*To mods: Please do not close this issue yet, thanks.

Sinchiguano, 2019-12-05:
Well, I used yumipy with kinetic and even melody, and I did not have problem at all. But there is another package that I think it is built on top of OrebroUniversity/yumi, here you have the link:
https://github.com/kth-ros-pkg/yumi
I tried with that one, I did couple of applications and it works pretty well, the difference between the yumipy and the kth-ros-pkg/yumi is that you will have more capabilities to leverage the yumi robot with MoveIt if you go with kth package.

Best regards
Cesar